sde1000/python-dali #2: Restructure the code

caiwan, 2015-08-18:
I'd create tickets to handle several things, we've been discussed - or started to - under the [previous pull request](https://github.com/sde1000/python-dali/pull/1)

I'll paste here your plan:

- dali/
	- exceptions.py
	- frame.py - forward and backward frames and transactions
	- command.py - command registry, interface to command decoding
- commands/ - commands and their responses
	- gear/ - commands to control gear
		- base.py - commands from part 102
		- fluorescent.py - commands from part 201
		- emergency.py - commands from part 202
		- incandescent.py - commands from part 205
		- led.py - commands from part 207
		- ... commands from other parts 203, 204, ...
	- device/ - commands to control devices and events from them
		- base.py - commands from part 103
		- ... commands from parts 301, 302...
- interface/
	- daliserver.py - interface to https://github.com/onitake/daliserver
	- tridonic.py - driver for Tridonic DALI-USB device
	- ... further interfaces

caiwan, 2015-08-18:
I've started this process in a separate branch, [caiwan:dev](https://github.com/caiwan/python-dali/tree/dev)

sde1000, 2015-08-18:
I see the first thing you did in this branch was merge from my master.  Can I suggest you rebase on my master instead?  That'll make the history look a lot more understandable, and give me the possibility of using your commits as they are without merging or cherry-picking.

caiwan, 2015-08-18:
I don't really know how to do so, that is the problem. I'm using git for years, and I'm still a newbie.

sde1000, 2015-08-18:
Hmm.  Looking at the history at the moment, I think it's probably best if you start again from my current master - it has all the changes you made apart from the __unicode__ / __str__ ones that we're not keeping, and which you got rid of when you merged from my master.

On the command line it'll be something like this (starting in your branch 'dev'):
```
git fetch upstream
git checkout -b newdev upstream/master
git cherry-pick dev
```
You are now in a branch "newdev" with your commit from dev applied on top of my current master.

You could then do:
```
git branch -d dev
git checkout -b dev
```
Then you can push it to github.  The first time you push you have to force it, because you're not pushing on top of what github currently sees as the dev branch.

sde1000, 2015-08-18:
I suggest you don't work on moving things around at the moment, though, because that will make the other changes that you make difficult to compare to my tree.

The "device.py" file definitely doesn't belong with the device driver and daliserver interface; ultimately it will go in its own subpackage along with "bus.py".

I'm not sure about "address.py" yet.  Some of the addressing concepts are shared between control gear and control devices, although the address spaces are separate; it may still make sense to have this shared between the two.

Before each commit, please would you check that the unit tests still pass with ```python -m unittest discover```.

rnixx, 2015-09-29:
I guess this issue can be closed in favour of #21

closed 2015-10-07 by sde1000