sde1000/python-dali #3: CommandTracking is completely broken

caiwan, 2015-08-19:
Having a metaclass to keep track of all class types, is not working. We'll have to find an other way to do it.

sde1000, 2015-08-19:
In what way is it broken?

caiwan, 2015-08-19:
The _commands property doesn't seem to exist, unless you intantitate these classes. I didn't found any solution, rather than registering these command types by hand.

sde1000, 2015-08-19:
Are you using python 3?  The way metaclasses are declared has changed between python 2 and python 3:

Python 2:

```
class foo(object):
    __metaclass__ = bar
    pass
```

Python 3:

```
class foo(metaclass=bar):
    pass
```

caiwan, 2015-08-19:
Is there any chance to make it work with version (at the same time)?

sde1000, 2015-08-19:
It's a bit of a pain; you either write it in quite a convoluted way, or make use of a compatibility library like [six](https://pythonhosted.org/six/).

How important is python 3 compatibility to you at the moment?  I think I mentioned before that I do intend to port this to python 3 at some point, but keeping python 2 compatibility is important to me for now; I make use of this library in some pubs that use [quicktill](https://github.com/sde1000/quicktill) which is python 2 for now.  (Although, having said that, it might be time for me to look at converting quicktill to python 3 now most of its library dependencies have been ported.)

caiwan, 2015-08-19:
I'll rollback my changes to see if it's broken under py 2.7 or not. I'll also get know that if py3 is really nesesarry, or not. (I'm fine w/ py2, but the environment, where it will be used... )

caiwan, 2015-08-19:
What exactly does the `_devicetype` parameter in the commands, and how it sohuld really work?

sde1000, 2015-08-19:
IEC 62386 defines many different types of control gear.  Some commands are common between all of them; these are defined in IEC 62386-102.  Other commands are specific to particular types of control gear, for example fluorescent ballasts, emergency lighting controllers, LED lighting controllers, and so on and these commands are defined in IEC 62386-201, 202, 203, and so on.  Part 201 defines commands for device type 0, part 202 defines commands for device type 1, and so on.

These device-specific commands (referred to in the standard as "application extended commands") all occupy the same "space" - they are encoded in ways that conflict with each other.  To issue one of these commands, you first send the EnableDeviceType(n) command with the appropriate value of n, so only devices of that type will respond to the command.

When you are decoding commands from their packed representation back to Command object instances, you need to specify the type of device the command is for so you get the right one.  If the previous command was EnableDeviceType(n) then the device type is n; if the previous command was anything else then the device type is 0.

rnixx, 2015-08-19:
if you want py2/3 compatibility (which actually is a good idea) with metaclasses, take a look at http://stackoverflow.com/questions/18513821/python-metaclass-understanding-the-with-metaclass

caiwan, 2015-09-09:
It seems ifex by now, I'll stay with using py 2.7

closed 2015-09-09 by caiwan