sde1000/python-dali #4: Package License

rnixx, 2015-09-04:
Hi,

would you mind license this package under LGPL, MIT, APACHE or similar license?

Regards

sde1000, 2015-09-04:
I might consider LGPL, but for a Python package I don't see what it gains you over GPLv3.  Can you explain what you want to do that's currently not possible?

rnixx, 2015-09-04:
I can't use your code in a non GPL project right now. IMHO GPL is good for complete software solutions, for libraries and frameworks i'd prefer less restrictive licensing.

sde1000, 2015-09-04:
I'm willing to add LGPLv3 to the licence.

Just a note, though - this library is still at a very early stage, and the API is neither complete nor stable.  Use with care, and be prepared for things to break when you pull an update.

rnixx, 2015-09-04:
Ok, sounds good. Prior to investigate deeper whether to use a piece of software or not i need to make sure i get no licensing issues.

Anyway, i'm going to try it out the next 2-3 weeks, and of course, i'm willing to make contributions in form of tests or whatever is necessary.

Thanks for your work!

rnixx, 2015-09-18:
How should the license change should be organized in code? Add LICENSE-LGPL file next to LICENSE? How should the changes be reflected in debian/copyright?

sde1000, 2015-09-18:
I followed (some of) the advice at http://www.gnu.org/licenses/gpl-howto.html

sde1000, 2015-09-18:
Ugh, sorry for the force-pushes.  I'm clearly not awake yet!

commit e0687750 (2015-09-18): Licence change: LGPL3+

A change of licence from GPL3+ to LGPL3+ was requested by a potential
user of the library.  This commit implements that change, and closes #4
on github.
commit f925388f (2015-09-18): Licence change: LGPL3+

A change of licence from GPL3+ to LGPL3+ was requested by a potential
user of the library.  This commit implements that change, and closes #4
on github.
commit 10ff0c1f (2015-09-18): Licence change: LGPL3+

A change of licence from GPL3+ to LGPL3+ was requested by a potential
user of the library.  This commit implements that change, and closes #4
on github.
closed 2015-09-18 by sde1000