sde1000/python-dali #10: dali.driver.tridonic.DaliUSB

rnixx, 2015-09-18:
Is this object experimental? legacy? The DaliServer class in dali.interface communicates with daliserver which actually does all the communication properly, so i am wondering if this class can go into a dedicated sandbox place if it's not supposed to end up as official API?

sde1000, 2015-09-18:
It's an experimental implementation of a driver for the Tridonic DALI-USB that doesn't require daliserver.  When it's complete it should support the same API as the driver in dali/interface.py.

rnixx, 2015-09-18:
ok, anyway, i pretty much like the approach with daliserver, it's slim and some might implement a control from a remote machine which has no physical access to the USB adapter.

closed 2015-09-18 by rnixx

sde1000, 2015-09-18:
Having a network protocol for DALI is definitely a good idea.  The one daliserver currently uses isn't sufficient in the long run, though, because it has no provision for the 24-bit frames used in IEC 62386 part 103.

rnixx, 2015-09-18:
Maybe someone adds support for 24 bit frames in the future. A ticket should be added at daliserver repo that anybody can take action.