sde1000/python-dali #12: PEP8

rnixx, 2015-09-18:
Do you mind if i format the code according to PEP8?

sde1000, 2015-09-18:
I thought @caiwan already did - but if there are parts he missed, please do reformat.  If you're making substantial changes, I suggest you rebase your tree on my current master so I can take your commit directly rather than having to cherry-pick it.

sde1000, 2015-09-18:
(It's just a git style thing - I dislike having branches/merges in the history that could trivially have been avoided if the author of a pull request had rebased their tree before generating the pull request.  I'm happy to have branches/merges for larger features or where a rebase wouldn't have been trivial.)

rnixx, 2015-09-18:
i've seen that @caiwan has been working on code restructuring, but it not seems there has anything been done related to PEP 8.

@caiwan - what's the state of your restructuring, is there any chance this can be merged soon?

Otherwise i would suggest formatting the code PEP 8 in the first place and do restructuring where appropriate afterwards... ?

caiwan, 2015-09-18:
Perhabs I have missed a few things when I was reformatting it. I was following Pycharm's guidance during the reformatting. 
I'll not reconstruct the code at the moment; I'll add the serial support soon.

rnixx, 2015-09-18:
@caiwan That means your formatting already has been merged into master?

caiwan, 2015-09-18:
Had been somewhere a month ago.

rnixx, 2015-09-18:
ok, thanks

caiwan, 2015-09-18:
I found it: [This commit](https://github.com/sde1000/python-dali/commit/40b3c27214ba0a308015358a9cbcd7672d7736b7)

sde1000, 2015-09-18:
A commit with formatting fixes from @rnixx has now been pushed to master.

closed 2015-09-18 by sde1000