sde1000/python-dali #9: Communicating with Tridonic Dali USB

rnixx, 2015-09-18:
As described in https://github.com/onitake/daliserver -> 4. Communication protocol, Requests are always 4 bytes long, thus the set-scene.py example fails (daliserver logs [2015-08-18 12:50:46] INFO Short read on connection 18, only got 2 bytes).

I digged around a little bit and added a hack to transmit protocol version and request type as expected here https://github.com/rnixx/python-dali/commit/27fa7f19b1c4de18dfd9c165f4f7af7f302c3a11

Am i missing any basic configuration for the Tridonic adapter or is this a bug?

sde1000, 2015-09-18:
It looks like this is a bug that was introduced during the restructuring that @caiwan contributed - he moved the "pack" operation from the driver to the Command class, and when one or the other of us updated the daliserver code we didn't spot this.  @caiwan - are you actually using the Tridonic device or just your own hardware?  (I have to admit I'm still using a much earlier version of the code with my Dali-USB devices, so I didn't catch this breaking - I don't have a suitable test environment for the Dali-USB at the moment because I've had to use my test device to replace a broken live one.  A long story involving flooding, wet electronics, and the DALI line-side of the device growing green furry corrosion!)

Anyway, the quick fix will be to pack the protocol version and request type in the driver and then append the packed bytes from the command.  I'll add a commit that does this.

The longer-term fix will be the "Frame" class I'm writing at the moment, instead of using tuples of integers to represent frames.

closed 2015-09-18 by sde1000

caiwan, 2015-09-18:
No, I'm not using tridonic, just only my controller through UART. The `pack` property will create variable length messege; If 32 bit padding is nesesarry, do add a `pack32` proprty, which fills the left space with zeroes (*or something*) I wasn't checked Tridonic USB for regression, sorry.

commit d4fe4060 (2015-09-18): Quick fix: add protocol version and request type in DaliServer

Oops - DaliServer wasn't sending the protocol version and request type
bytes when communicating with daliserver; it looks like this bug was
introduced when the "pack" operation was moved from the DaliServer code
to the Command object.

Fixes #9 on github, hopefully.