sde1000/python-dali #15: New 'Frame' class

sde1000, 2015-09-21:
I've started to add support for different frame sizes to the code.  Check out the 'frameclass' branch in my repo and let me know what you think.  In particular, I'd like to know whether this approach will work with what you've been doing in your own code.

Assuming you think this is OK, I plan to start going through IEC 62386-103 and adding 24-bit control device commands and events to the library.  There are also some more 16-bit commands in IEC 62386-102 to add, and some cleaning up to do to fix assumptions I made when I only had access to IEC 60929 annex E.

rnixx, 2015-09-21:
Looks very nice, but also a lots of changes. I'm stuck the next few days on project work, will try it out later this week.

But i guess this question was more in direction of @caiwan and his UART approach?

sde1000, 2015-09-21:
I have a busy week so I'm unlikely to be able to do any more before the weekend!

sde1000, 2015-09-22:
...well, I found some time.  I've reorganised to make the library match the structure of IEC 62386, added missing commands, and added incandescent lamp dimmer commands.

closed 2015-09-27 by sde1000