sde1000/python-dali #16: Driver Interface

caiwan, 2015-09-24:
Create an intrface class to make different drivers with it. (Sucha as tridonic USB, UART, DaliServer... etc) I've created one, but it made a merge conflict due I couldn't catch up the recent changes.

* Collect the common functions [like here](https://github.com/caiwan/python-dali/blob/dev-uart/dali/interface.py)
* Derive every class from it. Keep `interface.py` for contain interface for various drivers.

I've seen vou've created a new branch for serial communication, but I had my own [here](https://github.com/caiwan/python-dali/tree/dev-uart)

caiwan, 2015-09-24:
If anyone could poperly merge it, I'd appreciate that, since I've consideret I'd mess this up very quicly.

sde1000, 2015-09-24:
Yes, it needs tidying up - I'll look at it soon, and merge your driver too if that's OK with you.  The interfaces don't necessarily need to share a common ancestor class, but they do need to have similar APIs.  It's not yet clear what that API should be - they will ultimately need to work in multi-threaded and event-driven programs, so there needs to be room for experimentation.

The serial driver I just committed is for the Tridonic RS-232 DALI interfaces.  They appear to have a different protocol to the one you're currently writing a driver for - I had a quick look at yours to see if it matched, but it didn't.

caiwan, 2015-09-24:
I'd recommend to have a naming convention with branches:
- `master` is for all-time working code, with successful tests, etc
- `dev` is for collecting various features, tyding everyting up, make tests work
- `dev-<feature_name>` is for various features. 
I don't really know how to avoid conflicts like this, but keep somhow failsafe. (We need a poper workflow too)

sde1000, 2015-09-24:
Really not necessary with a distributed version control system.  Name branches in your tree whatever you like!  In my tree I will try to keep the "master" branch in a working state.

caiwan, 2015-09-28:
Would you take a look at my code?

sde1000, 2015-09-28:
I have rebased your UART driver onto my master branch; there were only minor merge conflicts.  You can find it in my branch "caiwan-dev-uart".  I haven't looked at the code beyond fixing merge conflicts and checking that the tests still run.

caiwan, 2015-09-29:
Thanks, I'll take a look at them, and create a pull request today.

closed 2015-09-29 by caiwan