sde1000/python-dali #18: Command tests takes ages

rnixx, 2015-09-27:
Running tests like so:

    ~/workspace/python-dali$ ./bin/python -m dali.test_command
    ....
    ----------------------------------------------------------------------
    Ran 4 tests in 85.095s

There's no USB DALI adapter connected while running the tests, so this might be a problem but then it would be better to have a backend mock for these tests.

sde1000, 2015-09-27:
The reason it takes a long time is because the strategy for testing is very basic.

The slowest test routine goes through each possible 16-bit DALI commands and makes sure it can be decoded into a Command object instance - and then takes that object instance and makes sure it encodes back to the same DALI command.

Then it does it for each of the first ten device types.

This has been very useful for getting full test coverage of all the subclasses of Command, and has caught many bugs.

An improvement would be to restrict the tests for device types other than zero to the portion of the DALI command space that they occupy.  This would probably speed up the tests by nearly ten times.

sde1000, 2015-09-27:
NB it may be possible to speed up the command tests further by restricting the test pattern even more.  The hard bit is doing this while keeping coverage.

commit bdda5f33 (2015-09-27): Speed and test improvements

Speed up testing by avoiding repeating tests for device types other than
0 on the part of the command space not covered by other device types

Remove abstract base classes from the list of classes used when decoding
frames.

Add a test to check that all concrete Command classes are covered by the
test pattern.

Closes #18 on github.
closed 2015-09-27 by sde1000