sde1000/python-dali #20: Using setuptools instead of distutils in setup.py

rnixx, 2015-09-27:
Is there a specific reason why you use plain distutils in setup.py instead of setuptools?

sde1000, 2015-09-27:
No particular reason, I started using distutils in my projects before setuptools existed and never got around to changing.

sde1000, 2015-09-27:
Also, I see that setuptools is an external dependency; it's still not part of the standard library.  I would need a compelling reason to change rather than a reason not to change!

rnixx, 2015-09-29:
According to https://python-packaging-user-guide.readthedocs.org/en/latest/current/ and the accepted answer at http://stackoverflow.com/questions/6344076/differences-between-distribute-distutils-setuptools-and-distutils2, setuptools is the most popular out there right now.

Actually i have to admit that I don't know the exact differences between these tools. I just use setuptools since ever so I wanted to know whether there's a reason to prefer distutils.

rnixx, 2015-09-29:
To sum this up, maybe just switch to setuptools once it's really necessary.

closed 2015-09-29 by rnixx