sde1000/python-dali #23: Interfaces to create various drivers for communication.

caiwan, 2015-10-01:
AS it turned out here, we'll need to specify interfaces to create cummunitation protocols and drivers with it - [here](https://github.com/sde1000/python-dali/pull/21).

rnixx, 2016-05-12:
I thought about how we can handle several DALI gateways and how a generic Driver might look in the end.

The general contract I am thinking about can be found here https://github.com/rnixx/python-dali/blob/driver_overhaul/dali/driver/base.py

The idea is that there is a first level of subclass implementing the protocol details, and a second level of subclasses which actually implement the sync or async read/write operations to the backend.

We can also solve it without subclassing, but it feels it not hurts here.

I also worked on the Tridonic DALI USB implementation (protocol details level) https://github.com/rnixx/python-dali/blob/driver_overhaul/dali/driver/tridonic.py

I can add the async second level of implementation as soon as missing parts expecially in TridonicDALIUSBDriver.receive are solved.

Ad comment from @sde1000 in #21

    What should the interfaces for these look like?
    Do we want an async interface with callbacks, like Twisted?
    Do we want a blocking interface as well, possibly built on top of it?

I'm voting for implementing this async based on the contract I suggested above. We can easily add another contract for sync communication based on it then.

For the backend IO related tasks we might want to add base communication objects based on pyusb and pyserial to ``dali.driver.base`` which can be used for the second level of implementation.

The code living in ``dali.interface`` should be adopted to the new contract and move then to ``dali.driver.daliserver``. Also the code in ``dali.driver.hasseb`` should be adopted.

The custom implementations of proprietary and non-open devices as pointed in #21 should be kept out from this package.

onitake, 2016-05-17:
Does it really make sense to separate drivers into asynchronous and synchronous interfaces?

While there are commands that expect a delayed response, the bus can only be used by a single sender at a time, and simple responses are always sent immediately after the command was received by the slave.

onitake, 2016-05-17:
And my 5¢ with respect to Twisted: I have never used it before and thus I don't know how it serialises data. But if the serialisation method is not user-definable, I'm against it. Language/framework lockdown is _always_ a bad idea. Better use something portable like REST or maybe Protobuf.

rnixx, 2016-05-17:
What do you mean with serialization method is not user definable? The recent implementation in PR #34 leaves the possibility for 3rd party library integration up to the user. Some 3rd party libs are async by design, thus I think it's a good thing to provide both sync and async base implementations.

onitake, 2016-05-17:
Sorry, I think I misunderstood how Twist works. I was under the assumption that it provides more than a network socket abstraction and event manager. If the protocol is defined in the application, this is obviously perfectly well.

rnixx, 2016-09-19:
@caiwan Hi, may you close this issue please?

closed 2016-09-19 by sde1000