sde1000/python-dali #25: Error when trying to run examples/find_ballasts.py

rnixx, 2015-10-06:
I tried to run examples/find_ballasts.py which ended up in an error.

I added ``print (command, result)`` before https://github.com/sde1000/python-dali/blob/master/dali/interface.py#L64

This is the output:

    ~/workspace/python-dali$ ./bin/python examples/find_ballasts.py 
    Usage: find-ballasts.py [host] [port]
    Looking for ballast devices at localhost:55825
    <dali.gear.general.Terminate object at 0x7f49f6cb3810> 
    <dali.gear.general.Initialise object at 0x7f49f6cb3810> 
    <dali.gear.general.Randomise object at 0x7f49f6cb3810> 
    Searching from 0 to 16777215...
    <dali.gear.general.SearchaddrH object at 0x7f49f6cb3810> 
    <dali.gear.general.SearchaddrM object at 0x7f49f6cb3810> 
    <dali.gear.general.SearchaddrL object at 0x7f49f6cb3810> 
    <dali.gear.general.Compare object at 0x7f49f6cb3810> �
    (<type 'exceptions.TypeError'>, TypeError(u'Response must be passed None or a BackwardFrame',), <traceback object at 0x7f49f6cc8d88>)
    Traceback (most recent call last):
      File "examples/find_ballasts.py", line 83, in <module>
        ballasts = find_ballasts(d)
      File "examples/find_ballasts.py", line 60, in find_ballasts
        low = find_next(i, low, high)
      File "examples/find_ballasts.py", line 41, in find_next
        response = i.send(Compare())
      File "/.../workspace/python-dali/local/lib/python2.7/site-packages/dali/interface.py", line 65, in send
        response = self.unpack_response(command, result)
      File "/.../workspace/python-dali/local/lib/python2.7/site-packages/dali/interface.py", line 90, in unpack_response
        response = command._response(rval)
      File "/.../workspace/python-dali/local/lib/python2.7/site-packages/dali/command.py", line 57, in __init__
        raise TypeError("Response must be passed None or a BackwardFrame")
    TypeError: Response must be passed None or a BackwardFrame

sde1000, 2015-10-06:
Ok, try commit 4c66c4552e0d718a0696ef6ab06c1fd57c832f7d.

rnixx, 2015-10-06:
Thanks! Works!

Found ballast at 10873224; withdrawing it...
Searching from 10873225 to 16777215...
(None, None, None)
8 ballasts found:
[10873217, 10873218, 10873219, 10873220, 10873221, 10873222, 10873223, 10873224]

closed 2015-10-06 by rnixx

sde1000, 2015-10-06:
Aha!  That looks like a single unit implementing eight DALI control gear devices - that's why the "random" numbers are a run.

If you want to assign short addresses to the gear, the code in dali/bus.py should be able to do that.

rnixx, 2015-10-06:
Yes, it's a 8-channel DALI dimmer.

I'm not yet into assignment, but thanks for the hint.