sde1000/python-dali #26: Confusing multiple_frames_per_connection kw arg in dali.interface.DaliServer

rnixx, 2015-10-07:
What is considered a "connection" here. A DaliServer instance? Because the socket connection gets opened and closed each enter/exit cycle if this flag is set to True, which is the opposite of multiple frames per (socket) connection in my understanding.

sde1000, 2015-10-07:
A TCP connection to daliserver.

If the flag is set to True then the connection is opened on ``__enter__`` and closed on ``__exit__`` and between those two multiple frames may be sent to daliserver.

If the flag is set to False then the connection is opened, a frame is sent and the response received, and the connection is closed; this happens for every frame.

The flag exists because sometimes daliserver gets into a state where it is "stuck" and requires killing and restarting if multiple frames are sent over the same TCP connection.  Unfortunately I don't have a test case that will reproduce this problem reliably, otherwise I'd have fixed the problem (or at least submitted a decent bug report) already.

rnixx, 2015-10-08:
Got it. I did not read the ``send`` function carefully enough.

> sometimes daliserver gets into a state where it is "stuck" and requires killing and restarting if multiple frames are sent over the same TCP connection

You need to restart daliserver itself? Or does the problem rely on the python side?

rnixx, 2015-10-08:
May you tell me what you have done to kill the server instance? Maybe I can at least reproduce the failure.

closed 2016-05-12 by rnixx