sde1000/python-dali #28: Writitng an async DALI USB Driver based on Twisted

rnixx, 2015-10-08:
@sde1000 as you mentioned in another post, dali.driver.tridonic.DaliUSB is meant as a ``daliserver`` replacement, with restriction that the DALI interface is not controllable via network any longer, but requires physical access to the underlying USB bus.

I plan to implement such a driver as Twisted based Protocol (twisted.internet.protocol.Protocol). A protocol has a ``transport`` property, where some can write data to and a ``dataReceived`` function, which gets called whenever data is received.

Some questions appear.

* The DaliServer interface waits for a response synchronous each time it sends a frame. Which might not be necessary for commands which not expect a response from the bus, except the DALI mouse excepts me to receive some kind of response. Do you know more about that?

* How should sending of frames which cause a response be handled in an asynchronous protocol in your opinion? One possibility would be to queue frames expecting a response and write the next frame once the response from the previous frame has been received and dispatched to a callback function given among the "request" frame. This would keep a request/response cycle sane, as long as no other master sends requests to the same bus, which might cause race conditions on the received responses. Non-response frames would not be a problem when received while a request is pending, would just be dispatched as regular input frame.

* Have you figured out reliably how a frame needs to look additionally to the real DALI payload for sending/receiving? I did not have a look at daliserver sources yet, the answer probably lives there in the code. sadly I'm not very familiar with C, so some help decrypting the interesting parts would be helpful.

* I would like to add the twisted based protocol code to python-dali. Even if someone not uses Twisted, the code will not make any problems as long as it does not get imported.

Thanks for you patience :)

sde1000, 2015-10-09:
The DALI-USB always replies to a command telling you what happened - including when the response is "there was no response", "there was a framing error in the received response", or "was unable to transmit because the bus is jammed".

Additionally it sometimes sends unsolicited data telling you about other traffic it observed on the bus.

The problem with writing a purely asynchronous driver for it is that pyusb does not have an asynchronous interface.  You will at a minimum need a thread reading from the device all the time, sending packets received from the device through a FIFO for the main event loop to pick up.

When you receive a response you always know what it is a response to: DALI slave devices have a very short window in which to respond if they are going to respond, so a master can always match up responses with transmitted commands.  You don't have to worry about race conditions there.  An asynchronous protocol should guarantee that there will be a callback after a forward frame has been transmitted and the status of the response is known, and these should always be received in order.

We don't yet know everything about what the commands sent to the DALI-USB look like.  Reverse-engineered documentation is in daliserver/lib/usb.c but only covers 16-bit forward frames.  More reverse-engineering work is needed to figure out how to send and receive 24-bit forward frames.

sde1000, 2015-10-09:
Thinking further on this: there are occasions where the master must be able to make a decision about what to transmit next in a very short time after receiving a response; for example, the documentation for "QUERY DEVICE TYPE" and "QUERY NEXT DEVICE TYPE" in part 102 sections 11.5.12 and 11.5.13 says that the sequence shall be transmitted as a transaction; this means that the master has 13.5ms to decide whether to send another QUERY NEXT DEVICE TYPE forward frame after seeing the response to the previous one.

This suggests that the application controller code must be fairly closely coupled with the driver code; it wouldn't make sense to put a network link between them, for example, that might have latency >10ms per round trip.  If we're going to introduce a network layer it should be slightly higher level than just "send forward frame / receive response" - although it should definitely support that too.

rnixx, 2015-10-09:
> If we're going to introduce a network layer

Which actually is daliserver, right?

You can use serial connections as transport with Twisted, no need to have a network connection/listener at all.

> You will at a minimum need a thread reading from the device all the time, sending packets received from the device through a FIFO for the main event loop to pick up.

This is what twisted does for you via services.

Latency needs to be measured in order to give a serious assessment on the approach.

For now I'm fine with the information that DALI-USB sends a response for each frame and is implemented as FIFO queue. So I can start experimenting.

> Additionally it sometimes sends unsolicited data telling you about other traffic it observed on the bus.

I can regulary listen to the whole bus with DALI-USB, right? I mean if I want to react to a Tridonic DALI XC or similar?

sde1000, 2015-10-09:
daliserver isn't the network layer that I think we want, ultimately.

In theory you can listen to the bus with DALI-USB and react to event frames (24-bit forward frames from control devices).  In practice we need to do more reverse-engineering work first to figure out how DALI-USB deals with 24-bit frames.

rnixx, 2015-10-09:
> daliserver isn't the network layer that I think we want, ultimately.

Right

> In practice we need to do more reverse-engineering work first to figure out how DALI-USB deals with 24-bit frames.

I'm at a conference next week, so I will have no time until 19th to dig deeper. Maybe I have some time reading frames sent by XC from the bus in the afternoon and write down the received packages somewhere (a ready to use test setup with XC, DALI-USB and an 8 channel dimmer is what i have)

sde1000, 2015-10-09:
We know what happens at the DALI level - it's defined in part 103.  We don't know what's happening at the USB level.  You'll need something that will dump the USB traffic between the DALI-USB and the Tridonic software.  There's a description of the setup @onitake uses [here](https://github.com/onitake/daliserver/issues/7).

rnixx, 2015-10-09:
Here is how and what I receive at DALI USB from Tridonic XC https://gist.github.com/rnixx/103862dcbe66ed966092

sde1000, 2015-10-09:
Right.  That looks like the Tridonic XC is sending 16-bit forward frames to control the gear directly, i.e. it has an embedded application controller.  See the fifth and sixth bytes of the received packet:
```
>>> from dali.frame import ForwardFrame
>>> from dali.command import from_frame
>>> import dali.gear.general
>>> unicode(from_frame(ForwardFrame(16,[255,0])))
u'Off(<broadcast>)'
>>> unicode(from_frame(ForwardFrame(16,[255,5])))
u'RecallMaxLevel(<broadcast>)'
>>> unicode(from_frame(ForwardFrame(16,[255,47])))
u'SetFadeRate(<broadcast>)'
>>> unicode(from_frame(ForwardFrame(16,[255,2])))
u'Down(<broadcast>)'
>>> unicode(from_frame(ForwardFrame(16,[255,1])))
u'Up(<broadcast>)'
```

What mode do you have the DALI-XC in?  I suspect it'll only send/receive 24-bit frames when it's in 'MC' mode.

rnixx, 2015-10-09:
It's already in MC mode. Position A
Other options are:
B: SC-A
C: SC
D: GC-A
E: GC
F: Prog
G: TuWh

sde1000, 2015-10-09:
Did you set it up to send those commands using the masterConfigurator software?

If it supports operating as a part 103 input device, I expect that'll happen when the DisableApplicationController command is sent to it.  Until we get 24-bit frames working, though, we don't have a way to send that command!

rnixx, 2015-10-09:
Yes, it's done with the masterConfigurator Software. Unfortunately the machine where it's installed is at another location right now and my local box runs Linux only. Is it possible to send DisableApplicationController via masterConfigurator?

sde1000, 2015-10-09:
I don't think I've seen that option.

If you have more than one DALI-USB then you could use one to listen to the DALI bus while the other is used with masterConfigurator to send configuration commands to the DALI XC - this might help figure out how DALI-USB reports 24-bit traffic.  It won't help figure out how to get DALI-USB to send 24-bit frames, though.

rnixx, 2015-10-09:
Do you know a cheap piece of hardware sending 24-bit frames? Right now I have no second DALI USB

sde1000, 2015-10-09:
I don't, at the moment.  At some point I will have to figure out my own system for USB traffic capture under Windows, and capture a log of masterConfigurator talking to the DALI-USB sending commands to the XC.

sde1000, 2015-10-09:
In the long run I plan to build my own DALI interface hardware (probably as a Raspberry "HAT"), but that's not very helpful right now...

rnixx, 2015-10-09:
OK, I'll purchase another DALI USB, so we can at least analyse the stuff sent from it.

Another question. Do you know whether I can send to the device from another thread while the receiver loop is waiting for a timeout?

sde1000, 2015-10-09:
They're not cheap - at least over here!

I'm pretty sure you can; that's the way I was planning to do it.

rnixx, 2015-10-09:
OK, Thanks for your help. I'll keep you updated with my findings. Need to leave now.

rnixx, 2015-10-20:
@sde1000

I just purchased another DALI USB. Tomorrow I'm going to play around a little with it.

Lunatone DALI 100k (http://lunatone.at/de/downloads/Lunatone_Art86458506_DALI_100k_Datasheet_EN.pdf) seems to be a device suitable for 24bit frame analysis.

From the docs:

Event  Message Mode

If a dimming value has to be transmitted, this mode enables the device to send a signal to a  specific target address (to itself or any other master), all in the scope of the proprietary protocol extension.

rnixx, 2015-10-21:
After adding a second DALI USB to my test setup and running "Addressing Wizard" from master configurator on the other device, this is what I receive https://gist.github.com/rnixx/aa317c05d6ce53cce0df

I printed out the 16 bit forward frames if byte 2 is 115.
Some packages have value 116 at byte 2, the received raw packet is printed then.

rnixx, 2015-10-21:
If you want to know the output of some particular action, please let me know

rnixx, 2016-05-12:
Hi, after some busy month, I have some time again to get back on this. I digged daliserver code a little bit and thought about how we can handle several DALI gateways and how a generic Driver might look in the end.

The general contract I am thinking about can be found here https://github.com/rnixx/python-dali/blob/driver_overhaul/dali/driver/base.py

The idea is that there is a first level of subclass implementing the protocol details, and a second level of subclasses which actually implement the sync or async read/write operations to the backend.

I also worked on the Tridonic DALI USB implementation here (First level of implementation) https://github.com/rnixx/python-dali/blob/driver_overhaul/dali/driver/tridonic.py

I have working async USB IO based on twisted on my disk, which could easily be made a non twisted version as well, but for now I need to know what frames need to be constructed in which circumstances in order to get further.

Would you be so kind and provide me some information related to
https://github.com/rnixx/python-dali/blob/driver_overhaul/dali/driver/tridonic.py#L70 + #L79 + #L96 + #L105 + #L112

I also would appreciate general feedback on the approach and whether this points the direction you imagined.

@sde1000 @caiwan  @onitake

rnixx, 2016-05-12:
I just seen that @caiwan opened issue #23 which is the right place for my last comment. I'll re-add it there and close this issue for now.

closed 2016-05-12 by rnixx