sde1000/python-dali #29: Accessing the collection of command types

caiwan, 2015-11-11:
There's a [metaclass](https://github.com/sde1000/python-dali/blob/master/dali/command.py#L9) that collects the known commands. Is there any method to access this and instantitate any class form it? 

I'm aiming to implement two features from it:
- There's a given raw frame, and some way I have to determine hat kind of frame it is, ad use the internal structures to disassemble it - simply reverse engineer it.
- The commands are given in string format, eg. `DAPC bcast 0x80` and the corresponding object has to be made, and sent. 

I'd implement these, and push my variations, if you could give me some hints how to do it properly to fit into the project.

*Note: Is there any way to classify this ticket as Enhancement?*

sde1000, 2015-11-11:
Going from a raw frame to a command is accomplished using dali.command.from_frame() - this uses the information collected by the metaclass.

Going from a string format command will require you to implement a parser for whatever string format you have decided on.  What's the grammar?  Who defined it, and where?  It doesn't really fit in with this library; the whole point here is to use Python classes for commands, and make use of Python's parser for generating DALI commands.

caiwan, 2015-11-26:
I've found what might be the problem. I was trying to use `CommandTracker.commands()`, but it doesn't work, because `CommandTracker` type doesn't have the `_commands` attribute.
This classmethod is gets `CommandTracker` type, as parameter, but `_commands` attribute is in class scope, so the type doesn't have it so. 

@slapec pointed it out, and gave me some hints to solve this:
- Initialize `cls._class_._commands` instead of `cls._commands` in the constructor of `CommandTracker` metaclass, so it would be the classmember of it, instead of `Command` (`cls == Command; cls.__class__ == CommandTracker`)
- Remove classmethod decorator from `CommandTracker.command`, so it would be intherited into `Command`, and the scope will matching.

I'll try if the first option will work, and do a pull request for it. This would solve all of my problems, because I can make a dict all of the known commands, and woul work for me as well. It doesn't have much business with this project, It's just a side problem I've bumped into, and requires this one to work.

sde1000, 2015-11-26:
Aha.  You introduced CommandTracker.commands() in your commit 3d81204d886c8cbe9aeea428ce2c129afe3c845d — but you should have made it be a class method of Command not of CommandTracker.  Just move it there and it'll be fine.

caiwan, 2015-11-26:
I've fixed it though.

closed 2015-11-26 by caiwan