sde1000/python-dali #35: Supported hardware list

Summer88, 2016-07-25:
Hi, 
This library project seems to be really interesting, but now I'm looking at hardware:

http://it.farnell.com/nxp/om13046-598/eval-dali-master-usb2dali/dp/2249920

is there anyone that have tried this board?

Are there any recommended usb2dali device?

Thanks, BR

sde1000, 2016-07-25:
I haven't tried that board.  I just had a quick look at its documentation.  It looks like it'll work for 16-bit forward frames and 8-bit reverse frames with the supplied firmware.  You could probably extend the firmware to deal with transmission and reception of 24-bit frames.  There should be enough information in its documentation and the supplied firmware source code to write a driver.

So far we've mostly been using the Tridonic/Lunatone DALI-USB device with the driver at https://github.com/onitake/daliserver

Summer88, 2016-08-05:
Good news,
received the board yesterday and is working fine with default firmware and hasseb.py driver.

Have a nice day!

closed 2017-03-19 by sde1000