sde1000/python-dali #36: find_ballasts.py

fabltd, 2016-09-14:
When running this module I get the following error:

TypeError: unsupported operand type(s) for >>: 'float' and 'int'

It appears that : 
low = 0
high = 0xffffff

if response.value is True:
        midpoint = (low + high) / 2

results in a value of 838607.5 (float)

Whats the fix?

rnixx, 2016-09-14:
I'd say

midpoint = round((low + high) / 2)

@sde1000 ?

fabltd, 2016-09-14:
Hi 

Its a python 3 issue:

In Python 3, / is float division

In Python 2, / is integer division (assuming int inputs)

In both 2 and 3, // is integer division

sde1000, 2016-09-14:
...and you can get the python 3 behaviour in python 2 using ```from __future__ import division```

python-dali doesn't support python 3 at the moment.  I will be fixing this before the end of November because I need python 3 support myself!

fabltd, 2016-09-14:
@sde1000 

Thanks. I have been modifying the code to run with Python 3 and an RS232 interface.

Do you expect the find ballast process to be quite slow?

sde1000, 2016-09-14:
Depending on the number of ballasts, yes it is slow.

dgomes, 2016-12-21:
Using python3 (trying to integrate DALI into home-assistant.io) and except for this all works in python3

sde1000, 2016-12-21:
Commit 27f890947bdfcfaa86bcdf8d6665469eb860e2b4 which I've just pushed may help with this.

If it doesn't, can you let me know how it's failing?  I don't have a test environment set up at the moment; I need to buy another DALI interface...

dgomes, 2016-12-21:
well I fixed it myself in the example code :) didn't notice there were more about it in the code.

Thanks for all your work!

closed 2017-03-19 by sde1000