sde1000/python-dali #42: No module named interface

wimg, 2017-11-05:
Whenever I try to run one of the examples, I get :

Traceback (most recent call last):
  File "find_ballasts.py", line 14, in <module>
    from dali.interface import DaliServer
ImportError: No module named interface

sde1000, 2017-11-05:
Aah.  It looks like that wasn't updated when things were moved around.

This is a bit of the project we're going to have to work on more, because at the moment you have to import a different thing depending on the hardware you're using.  If you're using daliserver you should change that import to ```from dali.driver.daliserver.DaliServer import DaliServer```.

rnixx, 2017-11-06:
> because at the moment you have to import a different thing depending on the hardware you're using

True, a factory function would be more appropriate here, but prior all drivers should match the new driver API.

closed 2020-05-08 by sde1000