sde1000/python-dali #45: Error when running find-ballast with SyncTridonicDALIUSBDriver

gabiSRC, 2018-06-06:
Hello,
I am currently trying to use the library in order to control my DALI ballast but I am in my very first steps.

I have a tridonic dali - usb adapter that is connected to my mac (macOS Sierra) and the adapter is not connected to any ballast yet as I want to make sure my environment is OK.

I adapted the find-ballast.py script in order to use the SyncTridonicDALIUSBDriver instead of Daliserver (although I am also trying to install and use Daliserver as a back-up solution if the native tridonic library does not work properly) but I get the following error when I execute it:
```
$ sudo python3 find-ballast.py 
Password:
Traceback (most recent call last):
  File "find-ballast.py", line 75, in <module>
    with SyncTridonicDALIUSBDriver() as d:
  File "/usr/local/lib/python3.6/site-packages/python_dali-0.6.dev0-py3.6.egg/dali/driver/tridonic.py", line 221, in __init__
  File "/usr/local/lib/python3.6/site-packages/python_dali-0.6.dev0-py3.6.egg/dali/driver/base.py", line 164, in __init__
  File "/usr/local/lib/python3.6/site-packages/usb/util.py", line 205, in claim_interface
    device._ctx.managed_claim_interface(device, interface)
  File "/usr/local/lib/python3.6/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

```

So I wonder if anyone has already experienced this and/or could help me with it.

One last weird thing is that even if my tridonic USB is connected to one of my few usb port, I cannot see any tty.USB in /dev/

Anyway, thank you for the help ;)

sde1000, 2018-06-07:
You wouldn't see a /dev/ttyUSB device with the Tridonic DALI-USB because it doesn't present as a serial interface over USB.

I think the question about permissions was answered by @onitake [in this issue](https://github.com/onitake/daliserver/issues/19)?

gabiSRC, 2018-06-08:
How is that possible that I do not see a ttyUSB? How can I identify the tridonic adapter then? I would like to identify that in order to pass it to a VM or a container.

And yes, as I thought, the permission issue have the same root cause as with daliserver :)

gabiSRC, 2018-06-08:
Oh, ok (sorry, I am not really not familiar with this)

Sorry if I ask (again) a dumb question: as I do not have tty, how can I pass it to a container? At first I wanted to do --device=/dev/ttyUSB0:/dev/ttyUSB0 but the path is not correct. Is there a way to work with daliserver or python-dali in a container then?

Thank you again for the help!

onitake, 2018-06-08:
Short answer: You can't.

Long answer: Your container needs to be able to access USB devices via sysfs/usbfs. If your container environment supports this, you can use the device. If it doesn't, you can't.

I suggest you try a VM with full USB pass-through support. The VM will run its own kernel and have its own (virtual) USB bus, where it will see the USBDALI adapter as a regular USB device, if it's passed through correctly.

gabiSRC, 2018-06-08:
I connected the dali adapter to a raspi3 and I executed the lsusb command
```
$lsusb
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 004: ID 17b5:0020 Lunatone 
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I do not see the adapter (maybe because it is not yet connected to any DALI device or power supply) but if it should appears here, I should have access to it in /dev/bus/usb/... I guess

sde1000, 2018-06-08:
It is the "Lunatone" one - they are the Original Equipment Manufacturer for the Tridonic DALI-USB.

gabiSRC, 2018-06-08:
Indeed, I missed the "17b5:0020" and was to focused on Tridonic

Thank you very very much!

gabiSRC, 2018-06-11:
Hello,
Ok it is good for me, using a docker container hosted on a raspi 3, I passed the usb bus to the docker and run in privileged mode, when I execute the daliserver all seems right:
```
$ daliserver
DALI USB multiplexer (daliserver)
Copyright (c) 2011 onitake All rights reserved.

[2018-05-11 08:18:24] INFO Starting daliserver
[2018-05-11 08:18:24] INFO Kernel driver is active, trying to detach
[2018-05-11 08:18:24] INFO Server ready, waiting for events
```
To pass the usb bus in the docker : --device="/dev/bus/usb/:/dev/bus/usb/" --privileged

Thank you for the help

closed 2018-06-11 by gabiSRC