sde1000/python-dali #48: Another DALI driver

rnixx, 2018-10-04:
Since there's some drive about DALI drivers at the moment i want to place my latest finding here as well

https://git.unipi.technology/UniPi/unipi-python-lighting/src/master/UnipiDali/unipidali.py

maybe it ends up in the driver collection somewhen

mhemeryck, 2019-06-05:
Had a similar question: if I would submit a PR for including it here, would you pull it in?

I so, I will also bring the ppl from [unipi](https://www.unipi.technology/) in the loop.

rnixx, 2019-06-05:
You might check whether license is ok (LGPL). Then i think there's no reason not to merge. @sde1000 ?

sde1000, 2019-06-05:
I'm away on holiday at the moment, I'll have a look next Tuesday.

mhemeryck, 2019-06-05:
No sweat, I'm only starting to look into DALI right now, might take some time to actually create the PR.

On a side note, I was also considering to prettify the code a bit: automated linting, pinning dependencies, ... How do you feel about this? Probably a different topic / issue ...

rnixx, 2019-06-05:
@mhemeryck please do not mix features an cleanup. btw - this package has no dependencies

mhemeryck, 2019-06-05:
I agree to not mix this feature with any possible cleanup -- I'll first look into the unipi driver (since I'll actually need that one quite soon).

On the dependency note: I did notice you were using the package `usb` somewhere, which isn't a standard python package. I assume it's [`pyusb`](https://pypi.org/project/pyusb/)? Anyways, I will try to get back to that in another topic like you suggested. I just already noted it since I couldn't run that project directly.

rnixx, 2019-06-05:
@mhemeryck ah right, yes. pyusb is needed for hasseb and tridonic drivers, my fault.

If you're further interested to contribute, there's another driver left to be adopted to the correct API (#54).

PR #59 (2019-10-10): Unipi axon s605 driver
Fixes #48 

Adds driver implementation for [Unipi Axon S605](https://www.unipi.technology/unipi-axon-s605-p179). My idea is to use this as part of my home automation setup. I would interface with this driver implementation by adding a [mqtt wrapper](https://pypi.org/project/paho-mqtt/) around it, converting events to commands on my DALI bus.

Implementation taken from https://git.unipi.technology/UniPi/unipi-python-lighting/commit/0975401ba6358d475ef46532ff5271bee46d601a adapted with unit tests. I did check the original code, and it does mention your licence; see https:/ [... cut]
closed 2019-10-10 by sde1000