sde1000/python-dali #52: Colour control gear support (device type 8)

jjonek, 2018-11-12:
I've used this library as glue between my lighting and other home automation and now that I have couple of couple of tunable white dt8 devices I want to add colour control into the play.

I acquired a copy of the part 209 and I'm currently working on writing `colour` module. For most of the things this looks quite straightforward, but there's a couple of things I'd like to evaluate my thoughts first. For instance, query colour value command takes parameter that defines which of the 50+ values is to be queried - would I rather define this as `class (int, Enum)` or just as constants?

sde1000, 2018-11-12:
I would definitely suggest some kind of enumeration class, because the library can be used to decode DALI commands as well as generate them; we will want to be able to go from the parameter value to its symbolic name.  (Although in this instance the parameter is passed by setting it in DTR0 first, so decoding in the general case could be tricky!)

ogerman, 2023-04-20:
If someone needs IEC 62386-209-2011 sheets, I can help to get access

jktjkt, 2023-04-20:
Just FYI, support for Tc (color temperature, aka tunable white) works just fine in the latest version (with some fixes in git). If I recall correctly there's also some code for other color LEDs (RGB, RGBW, etc), but I have not tested these because I do not have the HW.