sde1000/python-dali #53: Information

beireken, 2019-02-03:
I'm very interested in this project but i'm not exactly sure what i can do with this.
When i buy a hasseb usb adapter, can i use this to assign an adress to new balasts  (i guess i can do this with the hasseb software) and then switch / dim these?
How can i interface with the library (node-red, home-assistant)?
Is anyone using this with home-assistant or any other software to control their lighting?

dgomes, 2019-02-03:
I initially setup the addresses using the basses software, although you can use this library to achieve the same.

Personally I've used an Hasseb controller connected directly to my Home-Assistant (running on a RPi) setup using a custom-component: 
https://github.com/dgomes/home-assistant-custom-components/tree/master/light

and more recently I've moved to https://github.com/dgomes/dali2mqtt since I've went from using an RPi to using a VM for running my Home-Assistant setup.

beireken, 2019-02-03:
Ok, thx a lot! Good info!

rnixx, 2019-06-06:
This issue might be closed.

closed 2019-06-07 by beireken