sde1000/python-dali #56: Python 2 compatibility

sde1000, 2019-08-21:
Does anybody still care about python 2 compatibility for python-dali?

The next lot of code I plan to commit is python 3.5+ only, because it's based on asyncio.  Committing it won't break python 2 compatibility for existing code, but I'm just wondering about opportunities for clean-up.

rnixx, 2019-08-21:
I'm fine with dropping python 2 support.

dgomes, 2019-08-21:
I also support python3.5+

tobiasoort, 2019-08-21:
I just migrated my whole stack over to python 3.5 because that's the way the world works now. I'm fine with dropping py2.

mhemeryck, 2019-09-14:
Currently in the process of migrating our entire code base at work over the python3 (bit late of the party, I know). Let me know if you could use some help with that, figure it might be a nice opportunity for me to also learn a bit more of this repo.

sde1000, 2019-10-10:
Ok, the consensus appears to be that it's fine to drop support for python <3.5.  I'll remove it from the testing and packaging infrastructure soon, and after that there can be a gradual cleanup of python 2.7 compatibility code.  I'll leave this issue open until that cleanup is complete.

closed 2020-04-15 by sde1000