sde1000/python-dali #57: Hasseb driver throws (via libusb) USBError 110 Operation timed out

tobiasoort, 2019-08-21:
(I'm pretty sure this issue is on the device side of the USB-wire, but maybe someone knows something)

I've been developing with the Hasseb DALI device because ... well it was available and I want to control some lights in the house.

I can't get actual adressing to work, so all my commands are `DAPC(Broadcast(), <value>)` and `Off(Broadcast())` based. This shouldn't matter.

After implementing with an actual light, and generating some input for it, I noticed sometimes - even with seconds of 'nothing' around it - Python throws:

`  File "light.py", line 107, in send
    d.send(cmd)
  File "/usr/local/lib/python3.5/site-packages/python_dali-0.6-py3.5.egg/dali/driver/hasseb.py", line 85, in send
    self.backend.write(self.construct(command))
  File "/usr/local/lib/python3.5/site-packages/python_dali-0.6-py3.5.egg/dali/driver/base.py", line 189, in write
    return self._ep_write.write(data)
  File "/usr/local/lib/python3.5/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/usr/local/lib/python3.5/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/usr/local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 842, in intr_write
    timeout)
  File "/usr/local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/usr/local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
`

Does anybody know why this is? As far as I can check the line is 'free', no other stuff is using it, etc (how do I debug this?).

The hardware I'm working on is a Raspberry Pi Zero W, directly attached via USB to the Hasseb DALI device.

sde1000, 2019-08-21:
I had the same issue too.  I've had a hasseb device in my desk drawer for months, but I only started using it this week so I could test two masters on the same bus.  I wrote a completely new driver rather than try to fix the existing one!  (The device drivers in python-dali are rather a mess and need cleaning up.)

Would you like to try the new driver that I just committed?  You would need to pull from current master, copy `examples/50-dali-hid.rules` to `/etc/udev/rules.d/`, give udev a kick (plug/unplug the device, or reboot), then try `examples/async-flash.py` after editing it to change from `tridonic` to `hasseb`.

I'll see if I can add a script to do addressing using the new driver.

sde1000, 2019-08-21:
Ok, commit d378ef6d22df06a910ae148c051752eeb826498c includes a script to assign short addresses to control gear.

If you've added the udev rules, and assuming you only have one hasseb device attached, you can use the following command to assign short addresses:
```
examples/async-commissioning.py -d /dev/dali/hasseb-* -r hasseb --extend
```

closed 2020-04-16 by sde1000