sde1000/python-dali #62: How to deal with devicetype

ferdinandkeil, 2020-04-14:
I'm in the process of implementing a new driver (as mentioned in #59) and I stumbled over the device type issue. The CommandTracker (see #3) only stores these as far as I can see. [hid.py](https://github.com/sde1000/python-dali/blob/master/dali/driver/hid.py#L196) automatically deals with __devicetype_, but the other drivers don't.

If drivers are supposed to be handling the __devicetype_ attribute, it should be marked as public not protected.

If not, how should application logic handle it, e.g. interact with the CommandTracker?

sde1000, 2020-04-14:
I think you are right that it should be available as a public attribute on commands.  (This wasn't obvious when that code was written, but is now.)

So the question is: do we just rename _devicetype to devicetype throughout the code, or do we add a devicetype property that accesses _devicetype internally?

ferdinandkeil, 2020-04-14:
As a property would merely prevent changing the attribute, I think it would be more _pythonic_ to just rename it.

sde1000, 2020-04-14:
I'm inclined to rename it and add a property to enable it to be read as `_devicetype`, for compatibility.  I don't think there's much out-of-tree code that subclasses Commands, but there may be out-of-tree drivers already accessing `_devicetype`.

I'll wait a couple of days for any more comments, but if nobody really objects I'll go ahead and do this.

ferdinandkeil, 2020-04-14:
I also suggest adding a method to EnableDeviceType to create the correct command from a supplied command, e.g.:

```python
class EnableDeviceType(_SpecialCommand):
    """This command shall be sent before an application extended command.
    This command can be processed without the use of the Initialise()
    command.  This command shall not be used for device type 0.
    """
    _cmdval = 0xc1
    _hasparam = True

    @classmethod
    def from_command(cls, command):
        """Returns the correct EnableDeviceType command for the supplied command."""
        return EnableDeviceType(command.devicetype)
```

sde1000, 2020-04-14:
Good idea, although I would also extend it to raise a suitable exception if the command did not require an EnableDeviceType command (i.e. not 16-bit, or 16-bit and devicetype 0)

dgomes, 2020-04-14:
I propose the creation of the method `device_type` using the `@property` decorator.

sde1000, 2020-04-15:
Hmm.  I can see where you're coming from: similar name structure to the `is_config` and `is_query` properties.

I'm tempted to jump the opposite way, though, and rename all of `_appctrl`, `_inputdev`, `_uses_dtr0`, `_uses_dtr1`, `_uses_dtr2`, `_response` and `_sendtwice` to be explicitly public and be accessed as attributes rather than through properties.  With the exception of `is_config` and `is_query` I don't think this has come up yet because nobody has written code making use of them.

sde1000, 2020-04-15:
Regarding extending the `EnableDeviceType` class with a class method that takes another Command as an argument: there is at least one Command that this won't work for.  It is `dali.gear.general.QueryExtendedVersionNumber`.

ferdinandkeil, 2020-04-15:
> I'm tempted to jump the opposite way, though, and rename all of `_appctrl`, `_inputdev`, `_uses_dtr0`, `_uses_dtr1`, `_uses_dtr2`, `_response` and `_sendtwice` to be explicitly public and be [...]

👍 I support this.

Unfortunately, my copy of the standard is from 2010 and I can't find table 15 and 16. However, it does specify that commands in the configuration command range (32-129) or those dealing with assigning the short addresses (258, 259) need to be send twice. The latter - as far as I can see - do not belong to the category of configuration commands, defying the logic of `is_config`.

So I suggest:
- `is_config` and `is_query` get marked as deprecated.
- `expects_response` is added (same logic as `is_query`).

ferdinandkeil, 2020-04-15:
> Regarding extending the `EnableDeviceType` class with a class method that takes another Command as an argument: there is at least one Command that this won't work for. It is `dali.gear.general.QueryExtendedVersionNumber`.

Here the inconsistency in the DALI standard breaks the code. How about something along the lines of

```python
@property
def devicetype(self):
    raise RuntimeError('the device type for this command is not defined (see IEC 62386 Part 102)')
```

sde1000, 2020-04-15:
I'm going to solve this by mandating that all device type implementations provide their own version of `QueryExtendedVersionNumber`, using a mixin.  I'll update that pull request soon.

sde1000, 2020-04-15:
I merged the "rename devicetype" pull request.  I'll open another for renaming `sendtwice`, `uses_dtr0`, etc.

ferdinandkeil, 2020-04-16:
Awesome 👍

closed 2020-04-16 by ferdinandkeil