sde1000/python-dali #70: Driver and device discovery

sde1000, 2020-05-09:
The collection of device drivers isn't really "plug-and-play" at the moment.  All the example code imports the author's favourite driver and needs to be edited before being run to work with the others.  This has caused problems for people picking up the library for the first time (eg. #42, #44) and in any case is a bit tedious!

We need some kind of driver registry and device discovery module, so that simple scripts that just work with a single DALI device can call to get a suitable driver with the API they need.  Where multiple devices are present, or the device being used isn't automatically discoverable, the device to use could be specified by environment variable or on the command line.  More complex code that works with multiple devices at once also needs a way of naming devices (eg. in its config file or on the command line), so we need a way to go from a string description of a driver/device to a driver instance.

ferdinandkeil, 2020-05-11:
By device you mean the DALI interface?

sde1000, 2020-05-11:
Sorry, yes, that's unclear!  Yes, I mean the interface between whatever the code making use of python-dali is running on, and the physical DALI bus.

ferdinandkeil, 2020-05-12:
I think this should be solved by streamlining the driver API. At least for me, most of the confusion came from the huge differences in how the drivers are implemented.
Also, this sounds to me like you plan to come up with a general CLI for python-dali and that would probably better off as its own project.