sde1000/python-dali #82: async callbacks?

jschlyter, 2020-11-22:
Is there any way to create async callbacks, e.g., that the function provided to `register()` is async? Or is callbacks always sync?

cc @fredriklj

sde1000, 2020-11-22:
Are you talking about the code in `dali.drivers.hid`?  If so, callbacks are made using the `call_soon()` method of the event loop.  If you want to execute async code from the callback, you should be able to use `create_task()` in the normal way.

(Arguably the code that makes the callback ought to be able to spot that the target is a coroutine and call `create_task()` itself; I'll consider that for the future.)

jschlyter, 2020-11-22:
So something like this should work?

```
def callback(*args, **kwargs):
    await asyncio.create_task(async_callback(*args, **kwargs))
 ```

sde1000, 2020-11-22:
Except without the "await": you're creating a new task, not waiting for a coroutine to complete.

jschlyter, 2020-11-22:
Works, thanks a lot!

closed 2020-11-22 by jschlyter