sde1000/python-dali #84: Question about controlling +64 addresses

riparise, 2021-02-23:
Hello, I am new to the DALI protocoll, and I was hoping someone could help me with some ideas.

I want to controll a system that is made of 24 [drivers](https://www.watt24.com/media/65/86/0b/1592170311/Datasheet-Lunatone-Light-Management-LED-Dimmer-DALI-4Ch-LED-Dimmer-16A-89453832_1592170309_4.pdf) from a PC (linux or windows). Each driver has 4 individually addressable channels, (so in total 96 addressable channels). I am planning to use the Tridonic or Hasseb DALI master.

How can I overcome the 64 adresses limitation? I though of 2 solutions:

1) Will this library play well if I connect two masters to the computer?
2) Is there a way to reassing a short addresses every time I want to control a channel? Would there be any drawback?

Is there a better workaround?

Thanks

sde1000, 2021-02-23:
If you want to connect more than one interface to the computer, use the Tridonic/Lunatone ones: these devices have unique serial numbers so you'll be able to refer to them individually. The Hasseb devices don't have unique serial numbers so you won't be able to tell them apart. (The [example udev rules](https://github.com/sde1000/python-dali/blob/master/examples/50-dali-hid.rules) create a symlink for each Tridonic device based on its serial number.)

I don't think reassigning short addresses every time you want to control a channel will work. The process for assigning short addresses works at random (each device generates a random 24-bit address and the master works through those discovering them and assigning short addresses). The only things you can do with these 24-bit addresses are query and set short addresses. The 24-bit addresses are volatile and are reset on power-up.

rnixx, 2021-02-24:
To overcome the problem of identical serial numbers, you can use https://pypi.org/project/usbid/

With this package you can identify USB devices by path, which relates to the physical USB port on the computer.

closed 2021-09-23 by sde1000