sde1000/python-dali #87: Getting error ConnectionRefusedError: [Errno 111] Connection refused

AutobotZee, 2021-06-25:
I have downloaded and installed the library from the source and tried to run a test script "set_single.py" from examples. But after making all the changes and debugging some errors, the execution stops at send command and raises the error that the socket is unable to communicate to driver. The following error was shown. Will appreciate if someone can help me in debugging.

File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

sde1000, 2021-06-25:
It looks like the script is using `dali.interface.DaliServer` and wants to connect to daliserver — is that running? You can find it at https://github.com/onitake/daliserver

AutobotZee, 2021-06-25:
I installed the daliserver library and found out that process can not find the device. I followed the debian install instruction from @onitake but on starting the service it returns "*ERROR** Can't find USB device".

sde1000, 2021-06-25:
What does `lsusb` say? Can you see the interface there?

AutobotZee, 2021-06-25:
I can see the interface there listed. Its a Hasseb DALI USB, The listed name and buss address is 
Bus 001 Device 011: ID 04cc:0802 ST-Ericsson

sde1000, 2021-06-25:
Oh. Ok, daliserver is irrelevant then, it's only for Tridonic/Lunatone DALI USB. You will need to alter the script so it's making use of the Hasseb driver; see https://github.com/sde1000/python-dali/blob/master/dali/driver/hasseb.py

AutobotZee, 2021-06-25:
Hi Stephan,
Thank you for the important information
I tried to run the Hasseb.py driver and perform a test. I am pretty sure I am close now. 
I wrote a simple script to connect to the interface with the written driver using the forked repo [https://github.com/hasseb/python-dali]
```
#####
from dali.driver import hasseb
import dali.bus as bus

#Create hasseb USB DALI driver instance to handle messages
DALI_device = hasseb.SyncHassebDALIUSBDriver()
#Create DALI bus
DALI_bus = bus.Bus('hasseb DALI bus',DALI_device)
DALI_bus.initialize_bus()
####

```
In the hasseb.py file in line 82. the try code fails and as a result I get a null value in DALI_device.device_found in test code 
self.device = hidapi.hid_open(HASSEB_USB_VENDOR, HASSEB_USB_PRODUCT, None) 
[https://github.com/sde1000/python-dali/blob/master/dali/driver/hasseb.py]

When I tried to run an independent command with hidapi it still throws me error
```
Import hidapi
hidapi.init()
hidapi.hid_open(0x04cc,0x0802, None)

output:
 File "/usr/local/lib/python3.8/dist-packages/hidapi/hidapi.py", line 445, in hid_open
    raise RuntimeError('Could not open device.')
RuntimeError: Could not open device.
```

Could it be possible that I am missing some step to access the hasseb interface? As far as I know hidapi.hid_open() should just open the access to the interface.

sde1000, 2021-06-25:
That's probably a permissions issue. Did you copy https://github.com/sde1000/python-dali/blob/master/examples/50-dali-hid.rules to `/etc/udev/rules.d/` before plugging the device in?

sde1000, 2021-06-25:
Um. Also, is your clone of the python-dali repo up to date? `dali.bus` should no longer exist.

dgomes, 2021-06-25:
@AutobotZee probably my comment in this PR helps:

https://github.com/sde1000/python-dali/pull/78

AutobotZee, 2021-06-28:
The copying of rules helped. I also managed to install a more recent repo for [libusb/hidapi](https://github.com/libusb/hidapi) along with a better hidapi library mentioned by @dgomes. And following on the tips that it might be permission issue from @sde1000 i tried to execute my scripts with root privileges and made it work. Thanks for the info and help people. really appreciate it.

closed 2021-09-23 by sde1000