sde1000/python-dali #88: How to use the Commissioning sequence with SyncHassebDALIUSBDriver?

rousveiga, 2021-08-31:
Hello! I found the `Commissioning` sequence, and I would like to invoke it from my code. Thing is, I already instantiate some `SyncHassebDALIUSBDriver` objects, and I could only find examples of the sequence being run from a `hid`-type object.

My question is, how would I go about this? It doesn't seem that the driver class has an accessible `hid` (or `hasseb`) member - do I need to instantiate a `hasseb` object, or is there something in the `SyncHassebDALIUSBDriver` class that can run sequences?

Thanks in advance!

sde1000, 2021-08-31:
It looks like there's no sequence runner in that driver. A simple sequence runner would look something like this (untested, just from memory; I don't have an environment to test it in at the moment):

```
def run_sequence(driver, seq):
    response = None
    try:
        while True:
            try:
                cmd = seq.send(response)
            except StopIteration as r:
                return r.value
            if isinstance(cmd, dali.sequences.sleep):
                time.sleep(cmd.delay)
            elif isinstance(cmd, dali.sequences.progress):
                print(cmd) # or call something else to deal with it
            else:
                # XXX may need to send EnableDeviceType here for some commands
                # depending on whether or not the driver does it automatically
                response = driver.send(cmd)
    finally:
        seq.close()
```

rousveiga, 2021-08-31:
I see, I will try implementing it. Thank you for your help!

rousveiga, 2021-09-03:
I was going to make a PR to this repo but I forgot I had some other changes in my fork, adding support for use of more than one Hasseb driver - is it okay if I include those too in the PR?

sde1000, 2021-09-03:
Ideally one PR per feature if you can.

By the way, I had a quick look: the Hasseb driver doesn't send EnableDeviceType automatically when you send a command for a device type other than 0. So you'll need something like the following before the call to driver.send():

```
if cmd.devicetype != 0:
  driver.send(EnableDeviceType(cmd.devicetype))
```

rousveiga, 2021-09-06:
Okay, I'll test that. Thank you!

sde1000, 2021-09-23:
Is this sequence working for you now? If it is, I'll close the issue.

rousveiga, 2021-09-23:
Yes, it is! It runs perfectly with all four sequences. I will make a PR now.

closed 2021-09-23 by sde1000