sde1000/python-dali #95: Schedule for dropping support for Python 3.6

sde1000, 2021-11-27:
At some point it would be useful to drop support for Python 3.6, for two main reasons:

1. So we can make use of features introduced in Python 3.7.
2. To make support for async code easier. (Some asyncio calls required in Python 3.6 are deprecated in Python 3.8 and removed in later versions)

But when?  Security support for Python 3.6 ends in December 2021, five years after its initial release; there's an argument that says support can be dropped after that. However, some distributions (eg. Ubuntu 18.04) will continue supporting Python 3.6 for longer; support for Ubuntu 18.04 ends in April 2023, for example.

Current distribution Python versions are:
* Ubuntu 18.04: 3.6
* Ubuntu 20.04: 3.8
* Debian buster ("oldstable"): 3.7
* Debian bullseye ("stable"): 3.9
* RHEL 8: 3.6
* RHEL 8.2: 3.8
* Arch: 3.9 (but releases back to 3.4 are installable)

I would like to pick a date between 23rd December 2021 and end of April 2023!

dgomes, 2021-11-27:
In other projects the rule of thumb is support major distributions (Ubuntu, Debia, RHEL) stable/LTS versions

sde1000, 2022-06-27:
There is very little in the queue for release 0.10 so far, certainly nothing that current users of python 3.6 + release 0.9 would find useful. Continuing to support python 3.6 in future releases is becoming troublesome, especially for code that includes type annotations or makes use of asyncio.

Therefore I'm going to say that release 0.10 will drop support for python 3.6 and the minimum supported version will be python 3.7. It looks likely that release 0.10 will happen when the 24-bit support code from @sl-wallace is merged and stable.