sde1000/python-dali #98: Bug in querying scene levels

TobsA13, 2022-01-27:
Hi, I get a 0 as response when I querying a scene with has the value 255.
```
dali_driver = SyncHassebDALIUSBDriver()
a = address.Short(1)
b = gear.QuerySceneLevel(a, 8)
c = dali_driver.send(b)
d = c.value
print(d)

=> 0
```
This is a log from a Lunatone interface.
[image]

I tried to search the bug in the code, but I could not find it.

sde1000, 2022-01-27:
Do you get sensible responses when querying scenes that have values other than MASK?

TobsA13, 2022-01-28:
Yes other values are working correct. It's only the MASK value.

sde1000, 2022-01-28:
Hm. What's the value of `c.raw_value.as_integer`?

TobsA13, 2022-01-28:
Also 0

TobsA13, 2022-01-28:
Ok it seems that I have to query the value twice to get the right data.
So you are right, the issue is also on non "MASK" values.
```py
# Scene0 = 0, Scene1 = MASK, Scene2 = 246, Scene3 = MASK 

dali_driver = SyncHassebDALIUSBDriver()
a = address.Short(8)
print(dali_driver.send(gear.QuerySceneLevel(a, 0)).value) # MASK
print(dali_driver.send(gear.QuerySceneLevel(a, 0)).value) # 0
print(dali_driver.send(gear.QuerySceneLevel(a, 1)).value) # 0
print(dali_driver.send(gear.QuerySceneLevel(a, 1)).value) # MASK
print(dali_driver.send(gear.QuerySceneLevel(a, 2)).value) # MASK
print(dali_driver.send(gear.QuerySceneLevel(a, 2)).value) # 246
print(dali_driver.send(gear.QuerySceneLevel(a, 3)).value) # 246
print(dali_driver.send(gear.QuerySceneLevel(a, 3)).value) # MASK
```

TobsA13, 2022-01-28:
Ok the issue seems to occur when I have sniffing enabled. As soon I have sniffing disabled I get the right values.

I don't know if this is a bug, or you should not use sniffing and querying commands at the same time.

sde1000, 2022-01-29:
What's your setup?  You have a Hasseb device that you are controlling with your script, and a Lunatone DALI-USB also connected to the same bus for sniffing, running their software?

TobsA13, 2022-02-03:
HI sorry for the delay, wasn't home the last days.
Yes correct. The Hasseb device is my main device for the iot integration. The Lunatone device was for setting up some of their DALI hardware.

Do you use the sniffing of the Hasseb device? I also get many errors during the sniffing, so I can't use the output of the sniffing.
```
('ERROR', bytearray(b'\xaa\x07\x00\x06\x00\x00\x00\x00\x00\x00'))
('ERROR', bytearray(b'\xaa\x07\x01\x06\x01\x00\x00\x00\x00\x00'))
('ERROR', bytearray(b'\xaa\x07\x00\x06\x00\x00\x00\x00\x00\x00'))
```
Which should be:
[image]
My code for the sniffing, it does not matter if I use the Sync or the Async driver:
```python
from pprint import pprint
from threading import Thread
from dali.driver.hasseb import AsyncHassebDALIUSBDriver

dali_driver = AsyncHassebDALIUSBDriver()
dali_driver.enableSniffing()

def ref(driver):
    while True:
        data = driver.receive()
        if data is not None and len(data) > 0:
            text = ""
            if data[3] == 0x05:
                text += 'SNIFF'
            elif data[3] == 0x06:
                text += 'ERROR'
            else:
                text += 'OTHER'
            pprint((text, data))
            pprint((int(data[5]), int(data[6])))

thread = Thread(target = ref, args = (dali_driver, ))
thread.start()
```

sde1000, 2022-02-03:
I don't use the Hasseb device. I think you're going to have to talk to @hasseb directly about this; they wrote the firmware and supplied the driver.