sde1000/python-dali #100: As of python 3.10, the *loop* parameter was removed from Event() and BoundedSemaphore() of asyncio

fredriklj, 2022-04-23:
After upgrading my python environment I notices that the loop parameter has been removed from Event() and BoundedSemaphore(). Simply removing it seems to have fixed the issue, but I'm not sure on how to manage that inconsistency between versions.

sde1000, 2022-04-24:
It looks like it's been deprecated since Python 3.8 and is removed in 3.10; `Event()` (etc.) objects are now expected to use `asyncio.get_running_loop()` internally when necessary. This appears to tie into the [asyncio policies](https://docs.python.org/3.7/library/asyncio-policy.html) framework introduced in Python 3.7.

I think the correct thing to do here is to remove all the `loop` parameters from our use of asyncio, and from the next release of python-dali (0.10?) document that Python 3.6 is no longer officially supported, as discussed in #95.

sde1000, 2022-08-10:
This is fixed for the Tridonic DALI-USB driver in c72374c456b608c608ff65d2e076e4b49f734ba3

closed 2022-08-10 by sde1000