sde1000/python-dali #110: TypeError: __init__() got an unexpected keyword argument 'loop'

RastaKolobar, 2022-10-21:
Traceback (most recent call last):
  File "./async-flash.py", line 35, in <module>
    loop.run_until_complete(main(loop))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "./async-flash.py", line 11, in main
    dev = tridonic("/dev/dali/daliusb-*", glob=True, loop=loop)
  File "/home/sysadm/USB/venv/lib/python3.8/site-packages/python_dali-0.9-py3.8.egg/dali/driver/hid.py", line 337, in __init__
TypeError: __init__() got an unexpected keyword argument 'loop'

sde1000, 2022-10-21:
Will be fixed in release 0.10. The commit that fixes this is c72374c456b608c608ff65d2e076e4b49f734ba3

Release 0.10 should not be too far away. @sl-wallace — is there anything else you want to send over before I make the next release?

RastaKolobar, 2022-10-22:
Thanks Stefan!  I think that my environment is in sync with said commit - the error is still there.

sde1000, 2022-10-22:
Oh, good point — the driver was fixed but I didn't update the examples. I'll have a quick look at that.

sde1000, 2022-10-22:
Hopefully commit e856005bbc7b37c2ee32e1311d703774742e1fdf will work for you.

RastaKolobar, 2022-10-22:
Perfect ! Thank you, Stefan! Amazing project is python-dali. Now I will try to play with "memory" part of the library....

closed 2022-10-22 by sde1000

sl-wallace, 2022-10-23:
> Release 0.10 should not be too far away. @sl-wallace — is there anything else you want to send over before I make the next release?

I've got one more PR I've been working on, I'll send it over now

sl-wallace, 2022-10-23:
> I've got one more PR I've been working on, I'll send it over now

Just opened #111