sde1000/python-dali #120: Reseting all dali short address

ptorrent, 2023-06-06:
Hello there,

first at all, thanks for your awesome job !

I would like to know the sequence for removing all short addresses of a dali bus ?

is that the correct sequence ?
```
            yield DTR0(255)
            yield SetShortAddress(Broadcast())
```

Thanks for your help !

sde1000, 2023-06-06:
Yes, that should do it. (You can see it as part of the [Commissioning sequence](https://github.com/sde1000/python-dali/blob/6923d79cbb1c98f3503da006bff278e1cd673e77/dali/sequences.py#L175).)

ptorrent, 2023-06-06:
I did the test but it seems not working as I isolated the code for doing that..

What would be the Brocast() value for that ? 255 ?

ptorrent, 2023-06-06:
I've 2 dali light, everything is working fine with your python scripts. Now I try to clear their short address. So I isolated this code 

```
      yield DTR0(255)
      yield SetShortAddress(Broadcast())
```

But nothing seems happen, this is the sequence writed on dali bus

```
1x 0xa3 0xff
2x 0x7f 0x80
```

sde1000, 2023-06-06:
That should do it. The two sends of 0x7f 0x80 must occur within 100ms of each other, with nothing else being sent in between — are you managing this?

ptorrent, 2023-06-06:
I'm using lunaton USB DAlI2 device, 

i'm able to see both request:

```
0091.0001 0xa3 0xff
0091.0061 0x7f 0x80
0091.0076 0x7f 0x80
```

Only 15ms between 0x7f 0x80 . Is this normal ?

Are the command correct ? After sending thos commands, I'm still able to get ballastStatus if I target the short address

THanks for your help, that's realy appreciate !

sde1000, 2023-06-06:
They look correct, and 15ms between the two repetitions looks normal.

What do you get for `QueryControlGearPresent` on the short address? And if you broadcast `QueryMissingShortAddress` do you get a "yes" response?

sde1000, 2023-06-06:
Hang on, I might be being stupid. `SetShortAddress(Broadcast())` should be 0xff 0x80 not 0x7f 0x80!  0x7f 0x80 is `SetShortAddress(GearShort(63))`.

So you should look into how the wrong frame is being sent.

ptorrent, 2023-06-07:
Oh yes that was a test..

0xa3 0xff
0xff 0x80
0xff 0x80

But no changes :(

The funny fact is that i've only 2 ballasts and I can ask "ballasts status" on all addresses (means 20.21...40...63). They are all responding... How it's possible ? Do you've to call "Terminate" after ths reset or Initialise before ?

ptorrent, 2023-06-07:
Oh this is the case for all queries x)

Asking if ballast exists:

Request :  0x03 0x91 on short address 1
Answer :  47

Request :  0x03 0x91
Answer :  48

sde1000, 2023-06-07:
It sounds like you might have both units set to the same short address? Or there might be some other hardware issue interfering with communication.

What happens if you disconnect one of the units and only have one on the bus at a time?

ptorrent, 2023-06-08:
Hello !

Did you already test the LUBA USB ? Seems that status are a bit strange

[image]

[image]

Is it the same for rs232?