sde1000/python-dali #121: Device commisioning

ptorrent, 2023-06-08:
Hello !

There is an example for commisioning gears in sequence.py

Is it the same for Devices  but with device methods ?

sde1000, 2023-06-12:
It should be, although I haven't had a chance to test it.

ptorrent, 2023-10-28:
Hello there !

I'm desperately trying to address a short address to a device.

I just follow the sequence for gear and change call function to device : 

1) Terminate <Buffer 59 36 05 00 01 c1 00 00 f3>
2) Initialise <Buffer 59 36 05 00 81 c1 01 00 72>
3) Randomise  <Buffer 59 36 05 00 81 c1 02 00 71>
4) Write H <Buffer 59 36 05 00 01 c1 05 ff 09>
5) Write M <Buffer 59 36 05 00 01 c1 06 ff 0a> 
6) Write L  <Buffer 59 36 05 00 01 c1 07 ff 0b> 
Compare :  <Buffer 59 36 05 00 41 c1 03 00 b0> and I never receive answer from  that commande

Do you see something wrong about those commands ?

Thanks a lot for your support

ptorrent, 2023-11-10:
By listening lunatone macro commissionning, I saw that the initialise must have value of 127. Now you sequence works for dali2 addressing.

closed 2026-10-16 by sde1000