sde1000/python-dali #131: ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

TA1DB, 2023-11-04:
Hi,

I am using Hasseb Dali Master with it's GUI on Windows 11 without any difficulty.

When try find_ballasts.py on the same machine I get "ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it"

Could anyone give me a hand to resolve my this issue. Thanks in advance.

sde1000, 2023-11-04:
The `find_ballasts.py` example is written to use DaliServer. If you want to use it with a Hasseb device you need to edit it to use that driver instead.

The `find_ballasts.py` example isn't particularly useful — it doesn't do anything once it's found them. You may find the `async-commissioning.py` example more useful.

closed 2023-11-04 by sde1000

TA1DB, 2023-11-04:
Thanks. What should I enter as the device in Windows instead of this : " --device /dev/dali/hasseb-* "