sde1000/python-dali #134: ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'

Hadatko, 2024-02-02:
Hello i am starting to use this code to control my dali devices. I am getting this error (even the devices are found on bus) which looks like some error in python code?
ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'

sde1000, 2024-02-02:
We will at least need a stack trace to look at this problem.

Hadatko, 2024-02-02:
Sure, i will try provide as much information as i can. Do you have some doc related to debugging? Or any suggestion how to provide usefull data?

Hadatko, 2024-02-02:
Ok so this deeply :D

I want use https://github.com/dgomes/dali2mqtt project which is using your module. I connected this https://svitimprotebe.cz/eshop/dali-usb/ tridonic usb dali converter to my server and connected to it one dali device. 
Then i created docker container and i am trying to run mentioned project and i am getting errors which looks are related to your module.
```docker
FROM alpine
RUN apk add git python3 py3-pip py3-paho-mqtt py3-slugify py3-usb py3-yaml py3-text-unidecode py3-voluptuous py3-watchdog hidapi 
RUN pip3 install python-dali hidapi --break-system-packages
RUN git clone https://github.com/dgomes/dali2mqtt.git && cd dali2mqtt
ENTRYPOINT ["tail", "-f", "/dev/null"]
```
```bash
/dali2mqtt # python3 -m dali2mqtt.dali2mqtt
2024-02-02 19:29:08,209 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:08,209 DEBUG: Using <tridonic> driver
2024-02-02 19:29:08,227 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:08,230 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:08,305 DEBUG: Found lamp at address 2
2024-02-02 19:29:09,883 INFO: Found 1 lamps
2024-02-02 19:29:09,909 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
2024-02-02 19:29:15,910 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:15,913 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:15,972 DEBUG: Found lamp at address 2
2024-02-02 19:29:17,551 INFO: Found 1 lamps
2024-02-02 19:29:17,577 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
2024-02-02 19:29:20,578 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:20,580 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:20,639 DEBUG: Found lamp at address 2
2024-02-02 19:29:22,217 INFO: Found 1 lamps
2024-02-02 19:29:22,243 ERROR: AttributeError: 'TridonicDALIUSBNoResponse' object has no attribute 'value'
2024-02-02 19:29:31,244 DEBUG: Connecting to mqtt:1883
2024-02-02 19:29:31,247 ERROR: Could not load device names config <devices.yaml>, a new one will be created after successfull start
2024-02-02 19:29:31,306 DEBUG: Found lamp at address 2
2024-02-02 19:29:32,884 INFO: Found 1 lamps
```

sde1000, 2024-02-02:
You should open this issue in the dali2mqtt project — it looks like there's an issue with the way it deals with "no response" being returned by the driver.

closed 2024-02-02 by sde1000

Hadatko, 2024-02-02:
Understand. Thank you.

dgomes, 2024-02-02:
Well this is an issue that could better solved in python-dali

We have a base class for the driver, but errors, each driver has there own classes... we should have a common base class. 

Else daemons would need to account every single error object per driver.

rnixx, 2024-02-03:
@dgomes I agree. The driver code needs consolidation, see https://github.com/sde1000/python-dali/issues/69. I hope I soon have some time again to contribute.

Hadatko, 2024-02-03:
Guys i am just amazed that you all are still active and quick to respond ;)